ros-perception/image_pipeline #832: Trouble using image_proc package to rectify depth image for conversion to pointcloud

marcusvinicius178, 2023-07-26:
Hello, I am using ROS2 Galactic

I'm attempting to rectify depth image data from a Kinect sensor using the image_proc package in ROS2. The ultimate goal is to use this rectified image for conversion to a point cloud, which will then be used to populate an occupancy grid map.

However, I'm facing a problem where the image_proc node isn't producing any output, even though it seems to be correctly subscribed to the raw image topic.

[image]

Here is the launch file setup for my image_proc node:

```
#Rectify camera data to use as input of point cloud converter
image_rectify_node = Node(
    package='image_proc',
    executable='image_proc',
    name='image_proc',
    remappings=[
        ('in/image_raw', '/kinect_sensor/depth/image_raw'),
        ('out/image_rect', '/kinect_sensor/depth/image_rect'),
        ('in/camera_info', '/kinect_sensor/depth/camera_info')
    ],
    output='screen'
)
```

According to ros2 topic echo, the /kinect_sensor/depth/image_raw topic is being published correctly with depth data. However, no data appears on the /kinect_sensor/depth/image_rect topic.

The image_proc node parameters appear to be set to their defaults, and I've checked the depth image encoding to ensure it matches what image_proc is expecting.

A video of this behavior issue is available at this **link**: https://youtu.be/WjLjagE8M9E
Is there something I'm missing in my setup? Any insights would be greatly appreciated. Thank you.

marcusvinicius178, 2023-07-28:
Already fixed and answered here: 

https://answers.ros.org/question/417736/depth_image_to_laserscan-not-publishing-to-kinect_scan-topic-in-ros2-galactic/?answer=417845#post-id-417845

closed 2023-07-28 by marcusvinicius178

pranavkora, 2023-09-15:
Hi @marcusvinicius178, the attached link isn't working. Could you tell me how you fixed the issue? No data is being published from /scan topic. Similarly, when I use pointcloud_to_laserscan too, data is being published by /scan but I'm unable to visualize it on Rviz. I'm using ROS2 Foxy on Ubuntu 20.