ros-perception/image_pipeline #839: Remapping the output topic of image_publisher is not effective

meyerj, 2023-10-27:
If I run the following command with the binary installation of ROS Humble in Ubuntu 22.04:
```sh
$ ros2 run image_publisher image_publisher_node --ros-args -p filename:=logo.jpg -r image_raw:=foo
```
with a remapping of the [hard-coded default namespace `image_raw`](https://github.com/ros-perception/image_pipeline/blob/fcd39396274df164e926b906cae8dc929d16608f/image_publisher/src/image_publisher.cpp#L55) to some other name, `foo/image_raw`, the remapping only has an effect for the raw image topic, but not for the `camera_info`, `compressed` and `theora` topics and related parameters and services exposed by `image_transport`:

```sh
$ ros2 node info /ImagePublisher
/ImagePublisher
  Subscribers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /camera_info: sensor_msgs/msg/CameraInfo
    /foo/image_raw: sensor_msgs/msg/Image
    /image_raw/compressed: sensor_msgs/msg/CompressedImage
    /image_raw/compressedDepth: sensor_msgs/msg/CompressedImage
    /image_raw/theora: theora_image_transport/msg/Packet
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Service Servers:
    /ImagePublisher/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /ImagePublisher/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /ImagePublisher/get_parameters: rcl_interfaces/srv/GetParameters
    /ImagePublisher/list_parameters: rcl_interfaces/srv/ListParameters
    /ImagePublisher/set_parameters: rcl_interfaces/srv/SetParameters
    /ImagePublisher/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
  Service Clients:

  Action Servers:

  Action Clients:

```

Is there any other way to publish an image on an arbitrary topic with the `image_publisher` node?

That behavior may be related to https://github.com/ros-perception/image_common/issues/187 and affect all users of `image_transport::CameraPublisher`? If yes, this may also be the root cause of other issues, like https://github.com/ros-perception/image_pipeline/issues/824.

Furthermore the default node name, `ImagePublisher`, with capitals and not anonymized is unexpected, because it does not follow typical naming conventions (not sure whether there is a REP or other kind of specification on that).

mikeferguson, 2024-01-17:
This does appear to be be an issue in image_transport for all but the raw plugin... I'm still trying to figure out exactly where it's gong wrong (because the linked issue is actually been temporarily patched AFAICT)

mikeferguson, 2024-01-18:
Ok - so actually - it looks like we CAN fix this in image_pipeline - and actually, we should fix it here - by passing the fully remapped name as the base name. PR that fixes just this node: https://github.com/ros-perception/image_pipeline/pull/851

closed 2024-01-19 by mikeferguson