ros-perception/image_pipeline #918: fatal error: cv_bridge/cv_bridge.hpp in humble branch

ShaharSarShalom, 2024-01-29:
hi 

In humble branch cv_bridge.hpp is not where to be found. 

while compiling (colcon) the image_publisher package, cv_bridge.hpp is not found. 

I'm getting this error: 

> colcon build --packages-select image_publisher  --allow-overriding image_publisher  --symlink-install --parallel-workers 4 --ament-cmake-args -DPYTHON_EXECUTABLE="/home/robo01/anaconda3/envs/CondaEnvForRos310/bin/python"  -DCMAKE_INCLUDE_PATH="/opt/ros/humble/include/cv_bridge/" -DCMAKE_BUILD_TYPE=Release
Starting >>> image_publisher
--- stderr: image_publisher                             
/home/robo01/PycharmProjects/temp_ros2_ws/src/image_pipeline/image_publisher/src/image_publisher.cpp:39:10: fatal error: cv_bridge/cv_bridge.hpp: No such file or directory
   39 | #include "cv_bridge/cv_bridge.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/image_publisher.dir/build.make:76: CMakeFiles/image_publisher.dir/src/image_publisher.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/image_publisher.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< image_publisher [0.35s, exited with code 2]

Summary: 0 packages finished [1.28s]
  1 package failed: image_publisher
  1 package had stderr output: image_publisher

Solve this issue by: 

In image_publisher.cpp, line 39 

chagne from: 
`#include "cv_bridge/cv_bridge.h"`
to 
`#include "cv_bridge/cv_bridge.hpp"`

mikeferguson, 2024-01-29:
What is your setup here? Because this is the correct file if using the Debian packages:

 * http://repo.ros2.org/status_page/ros_humble_default.html reports that cv_bridge is version 3.2.1 in humble
 * 3.2.1 was released 2 years ago, and includes ONLY cv_bridge.h (no .hpp file)

mikeferguson, 2024-01-29:
I think you have the wrong branch checked out - the "humble" branch of image_pipeline does NOT use the .hpp files: https://github.com/ros-perception/image_pipeline/blob/humble/image_publisher/src/image_publisher.cpp#L39

CI is also passing, both for source builds of humble, as well as the Debian package builds.

ShaharSarShalom, 2024-01-30:
Thank you so much, you are absolutly right 

I did - 

`git checkout -B humble`

instead of doing 

`git checkout -B humble origin/humble`

thank you so much

closed 2024-01-30 by ShaharSarShalom