ros-perception/image_pipeline #924: range_max sensor value introduced but not used

S-A-M-J, 2024-02-01:
I was wondering why in point_cloud_xyz.cpp in line 112 convertDepth is called but there is no way to set the max_sensor range that is part of the function based on a parameter given to the nodelet. I mean why implement the max sensor range variable and then not make it usable?

The option to set this parameter when calling convertDepth is not used in a single nodelet definition and would be a really useful tool for map generation. Perhaps someone can shed some light on that choice.

mikeferguson, 2024-02-02:
I'm thinking this is some sort of oversight, but it's likely nobody knows why at this point.

A PR that adds a parameter to enable this behavior would be accepted (we shouldn't change the default behavior).

PhilippPolterauer, 2024-02-17:
I tried tackling this issue with the PR #943. I would highly appreciate if you could have a look, and discuss with me some further steps.

PR #943 (2024-02-21): 924 introduce default depth param
this pull request attempts to fix #924 

changes:
- [x] changed the argument name `range_max` which was missleading to `invalid_depth` which captures the true meaning of the paramter
- [x] Introduction of a parameter called `invalid_depth`, which is the value that will be assigned to points that are not valid according to `DepthTraits<T>::valid`
- [x] small fixes regarding use of non const shared ptr references
- [x] changed for loop from int to uint32_t avoiding many type casts

Open:

- [ ] UB when reinterpret casting the pointcloud data? 
- [ ] introduction [... cut]
closed 2024-02-21 by mikeferguson