ros-perception/image_pipeline #949: ERROR Building in Humble: error: ‘struct rclcpp::PublisherEventCallbacks’ has no member named ‘matched_callback’

PlayWeird, 2024-02-23:
I'm trying to build your image_pipeline from source but keep getting this error. Any help would be appreciated. I'm currently using ROS2 Humble. 

```
Starting >>> tracetools_image_pipeline
Starting >>> camera_calibration
Starting >>> depth_image_proc
Starting >>> image_publisher
Starting >>> image_rotate
Starting >>> image_view                                                                             
Finished <<< tracetools_image_pipeline [0.56s]                                                                 
Starting >>> image_proc
Finished <<< image_publisher [0.93s]                                                                 
Finished <<< image_view [1.02s]                                                                      
--- stderr: camera_calibration                                                                       
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< camera_calibration [1.19s]
--- stderr: image_proc                                                                                   
/home/gaetano/image_pipeline/src/image_pipeline/image_proc/src/rectify.cpp:37:33: warning: extra tokens at end of #include directive
   37 | #include "cv_bridge/cv_bridge.h"#include "tracetools_image_pipeline/tracetools.h"
      |                                 ^
/home/gaetano/image_pipeline/src/image_pipeline/image_proc/src/rectify.cpp: In constructor ‘image_proc::RectifyNode::RectifyNode(const rclcpp::NodeOptions&)’:
/home/gaetano/image_pipeline/src/image_pipeline/image_proc/src/rectify.cpp:67:31: error: ‘struct rclcpp::PublisherEventCallbacks’ has no member named ‘matched_callback’
   67 |   pub_options.event_callbacks.matched_callback =
      |                               ^~~~~~~~~~~~~~~~
/home/gaetano/image_pipeline/src/image_pipeline/image_proc/src/rectify.cpp:68:12: error: ‘rclcpp::MatchedInfo’ has not been declared
   68 |     [this](rclcpp::MatchedInfo &)
      |            ^~~~~~
/home/gaetano/image_pipeline/src/image_pipeline/image_proc/src/rectify.cpp:88:48: error: too many arguments to function ‘image_transport::Publisher image_transport::create_publisher(rclcpp::Node*, const string&, rmw_qos_profile_t)’
   88 |   pub_rect_ = image_transport::create_publisher(this, "image_rect", qos_profile, pub_options);
      |               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/gaetano/image_pipeline/src/image_pipeline/image_proc/include/image_proc/rectify.hpp:41,
                 from /home/gaetano/image_pipeline/src/image_pipeline/image_proc/src/rectify.cpp:39:
/opt/ros/humble/include/image_transport/image_transport/image_transport.hpp:53:11: note: declared here
   53 | Publisher create_publisher(
      |           ^~~~~~~~~~~~~~~~
In file included from /opt/ros/humble/include/rclcpp/rclcpp/service.hpp:34,
                 from /opt/ros/humble/include/rclcpp/rclcpp/callback_group.hpp:28,
                 from /opt/ros/humble/include/rclcpp/rclcpp/node.hpp:40,
                 from /opt/ros/humble/include/image_transport/image_transport/image_transport.hpp:37,
                 from /home/gaetano/image_pipeline/src/image_pipeline/image_proc/include/image_proc/rectify.hpp:41,
                 from /home/gaetano/image_pipeline/src/image_pipeline/image_proc/src/rectify.cpp:39:
/home/gaetano/image_pipeline/src/image_pipeline/image_proc/src/rectify.cpp: In member function ‘void image_proc::RectifyNode::imageCb(const ConstSharedPtr&, const ConstSharedPtr&)’:
/home/gaetano/image_pipeline/src/image_pipeline/image_proc/src/rectify.cpp:95:3: error: ‘ros_trace_image_proc_rectify_init’ was not declared in this scope
   95 |   TRACEPOINT(
      |   ^~~~~~~~~~
/home/gaetano/image_pipeline/src/image_pipeline/image_proc/src/rectify.cpp:102:5: error: ‘ros_trace_image_proc_rectify_fini’ was not declared in this scope
  102 |     TRACEPOINT(
      |     ^~~~~~~~~~
/home/gaetano/image_pipeline/src/image_pipeline/image_proc/src/rectify.cpp:116:5: error: ‘ros_trace_image_proc_rectify_fini’ was not declared in this scope
  116 |     TRACEPOINT(
      |     ^~~~~~~~~~
/home/gaetano/image_pipeline/src/image_pipeline/image_proc/src/rectify.cpp:137:5: error: ‘ros_trace_image_proc_rectify_fini’ was not declared in this scope
  137 |     TRACEPOINT(
      |     ^~~~~~~~~~
/home/gaetano/image_pipeline/src/image_pipeline/image_proc/src/rectify.cpp:160:3: error: ‘ros_trace_image_proc_rectify_fini’ was not declared in this scope
  160 |   TRACEPOINT(
      |   ^~~~~~~~~~
gmake[2]: *** [CMakeFiles/rectify.dir/build.make:76: CMakeFiles/rectify.dir/src/rectify.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:209: CMakeFiles/rectify.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< image_proc [3.63s, exited with code 2]
Aborted  <<< image_rotate [5.17s]                                                                               
Aborted  <<< depth_image_proc [8.60s]                               

Summary: 4 packages finished [9.19s]
  1 package failed: image_proc
  2 packages aborted: depth_image_proc image_rotate
  4 packages had stderr output: camera_calibration depth_image_proc image_proc image_rotate
  2 packages not processed

```

mikeferguson, 2024-02-24:
@samuel-ongzx is correct here - you need to use the correct branch - humble branch is passing CI so there should be no issue once you switch to that.

closed 2024-02-24 by mikeferguson

PlayWeird, 2024-02-24:
I checked and you were correct. My issue was that I was switching to the humble branch on the GitHub page thinking that the HTTPS link was also switching to that branch. Thanks.