ros-perception/image_pipeline #950: Are the Realsense packages needed if a Realsense camera is not being used?

PlayWeird, 2024-02-24:
I'm using the Humble branch and get an error that I need to install the ros2_intel_realsense package (https://github.com/intel/ros2_intel_realsense)

```
[INFO] [launch]: All log files can be found below /home/gaetano/.ros/log/2024-02-24-07-52-38-817593-savage-jelly-11761
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'realsense_ros2_camera' not found, searching: ['/home/gaetano/image_pipeline/install/image_pipeline', '/home/gaetano/image_pipeline/install/stereo_image_proc', '/home/gaetano/image_pipeline/install/image_proc', '/home/gaetano/image_pipeline/install/tracetools_image_pipeline', '/home/gaetano/image_pipeline/install/image_view', '/home/gaetano/image_pipeline/install/image_rotate', '/home/gaetano/image_pipeline/install/image_publisher', '/home/gaetano/image_pipeline/install/depth_image_proc', '/home/gaetano/image_pipeline/install/camera_calibration', '/opt/ros/humble']"

```

Is this needed if I am not using a Realsense camera or can I just comment out the launch_ros.actions.Node in the launch file?

mikeferguson, 2024-02-28:
Those launch files do use real sense - but nothing in image_pipeline specifically requires it.

Those files are largely left over testing files from the days when Realsense was one of the only camera sensors that had ROS2 drivers - I imagine we will either clean up or remove the launch files at some point (some of them don't even work - it makes no sense to have realsense camera images go into a radial point cloud node, since they don't generate radial images)

PlayWeird, 2024-02-28:
Thanks for this, I've been struggling with these launch files but will move to running the nodes individually.