ros-perception/image_pipeline #955: [depth_image_proc]  RGBA8 support dosen't work with Humble package 3.0.3

macmacal, 2024-02-29:
Hello,
I'm experiencing a bug related to RGBA8 support in `point_cloud_xyzrgb_node` after upgrading `ros-humble-depth-image-proc` package from `amd64 3.0.0-1jammy.20240126.002447` to `amd64 3.0.3-1jammy.20240217.083751`.

Without any change (other than upgrading the packages) to my code base with simulated depth camera (in O3DE ROS 2 Gem) I started receiving the following error:
```bash
[component_container-6] [ERROR] [1709211425.531158009] [point_cloud_xyzrgb_node]: RGB image has unsupported encoding [rgba8]
```
(Logged from [this code line](https://github.com/ros-perception/image_pipeline/blob/551bc00aa9deab54b2c2ec36bd73d145219bea18/depth_image_proc/src/point_cloud_xyzrgb.cpp#L235))

Downloading `humble` branch of this repository and manually building it resolves the problem. Probably something went wrong with the binaries.

mikeferguson, 2024-03-02:
It looks like there is one additional commit related to the RGBA8 support - that isn't released in 3.0.3 - https://github.com/ros-perception/image_pipeline/commit/551bc00aa9deab54b2c2ec36bd73d145219bea18 - so we need a new release

mikeferguson, 2024-03-02:
3.0.4 tagged and released - will be some time before it makes it into a sync on the build farm: https://github.com/ros/rosdistro/pull/40046

closed 2024-03-02 by mikeferguson