ros-perception/image_pipeline #956: Add documentation for additional parameters of depth_image_proc::RegisterNode

Alessio-Parmeggiani, 2024-02-29:
Hi, I noticed that the parameter `fill_upsampling_holes` and `use_rgb_timestamp` of the register node from depth_image_proc are not specified in the [Documentation page](http://docs.ros.org/en/rolling/p/depth_image_proc/components.html#depth-image-proc-registernode).

They could be added with the documentation with the text:

- use_rgb_timestamp (bool, default: false) : use timestamp of rgb image instead of depth image for the registered image.
- fill_upsampling_holes (bool, default: false) : when RGB is higher res, interpolate by rasterizing depth triangles onto the registered image.

ahcorde, 2024-02-29:
@Alessio-Parmeggiani do you mind to open a PR ?

ahcorde, 2024-03-01:
All good, Thank you for your contribution!

closed 2024-03-01 by ahcorde