ros-perception/image_pipeline #964: point_cloud_xyzrgb_node errors:  RGB image has unsupported encoding [bgra8]

Alessio-Parmeggiani, 2024-04-02:
Hi, I got an error while running pointcloudXYZRGB node, it says the following error and it does not produce any pointcloud:
`[ERROR] [point_cloud_xyzrgb_node]: RGB image has unsupported encoding [bgra8]`

But looking in the code it seems that bgra8 is a supported format: [code](https://github.com/ros-perception/image_pipeline/blob/bae2fb952629d559b915c998e124f0b7643efc19/depth_image_proc/src/point_cloud_xyzrgb.cpp#L270)

The image is indeed of type bgra8, the following is one message from the camera:
```
header:
  stamp:
    sec: 1711115480
    nanosec: 428550000
  frame_id: zedxm_left_camera_optical_frame
height: 600
width: 960
encoding: bgra8
is_bigendian: 0
step: 3840
data: '<sequence type: uint8, length: 2304000>'
```

How can I solve this problem?

mikeferguson, 2024-04-02:
Those lines/fixes are only included in 4.0.1 - which was only released last week and has not yet made it into a sync. You can either switch to the "testing" Debian sources, build from source, or wait for the sync

Alessio-Parmeggiani, 2024-04-03:
@mikeferguson Ok thank you, do you know approximately when the sync will happen?

mikeferguson, 2024-04-03:
That's entirely up to the ROS boss for iron - typically happens on a 2-4 week cadence

closed 2024-04-03 by mikeferguson