ros-perception/image_pipeline #965: image_publisher_node not read camera_info_url on initialization

Kotochleb, 2024-04-17:
As in the title, `image_publisher_node` doesn't account for `camera_info_url` on the initialization when launched as a standalone node. For some reason, it skips reading it. When the same parameter is changed dynamically after the node started, everything works just fine.
The simplest walk-round I found was to change `return result` with `breake;` in the `pram_change_callback`. Yet, I feel like there should be a better solution to this, and I do not feel confident enough with the original implementation to submit a good enough bugfix PR.

mikeferguson, 2024-04-23:
I'm not sure why there is a "return result" in any of those cases - I would think we should iterate through all the parameter settings by completing the for loop (and then there is a return after the reconfigure callback is called). I think we should actually remove the lines entirely (not change to a break).

Kotochleb, 2024-04-24:
> I'm not sure why there is a "return result" in any of those cases

@mikeferguson this is precisely what caused my confusion and said lack of said confidence in making changes. If you think removing the `return` entirely is a fine fix I can submit PR soon.

mikeferguson, 2024-05-30:
Looks like fix is merged to Rolling, backports in progress

closed 2024-05-30 by mikeferguson