ros-perception/image_pipeline #966: camera_calibration tests call deprecated image_geometry methods

ScottMonaghan, 2024-04-20:
This issue is a follow-up to 
- #572 

Following the thread to fix #572 above, we [overhauled the `image_geometry` package](https://github.com/ros-perception/vision_opencv/pull/527) to both to internally deprecate any use of the np.matrix class, and also to bring the package into compliance with ROS 2 [code standards](https://docs.ros.org/en/rolling/The-ROS2-Project/Contributing/Code-Style-Language-Versions.html#id4).

Those changes introduced the deprecation warnings referencing `fromCameraInfo()` and `projectPixelTo3d()` below.

I'll go ahead and start working on a fix for these.

Here is my MUST, SHOULD, OUT OF SCOPE acceptance criteria for changes I think are appropriate. 

@vrabaud, @JWhitleyWork, @jacobperron, @mikeferguson, please let me know if you'd like me to take a look at anything else while I'm in there.

#### MUST HAVE
- [ ] camera_calibration test does not raise deprecation warnings

#### SHOULD HAVE
- [ ] camera_calibration python scripts do not reference any deprecated `image_geometry` members
 
#### OUT OF SCOPE
 - disutils package warning below (this should be a separate issue)

============================= warnings summary ===============================
src/camera_calibration/calibrator.py:47
  Warning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives

test/test_directed.py::TestDirected::test_stereo
test/test_directed.py::TestDirected::test_stereo
test/test_directed.py::TestDirected::test_stereo
test/test_directed.py::TestDirected::test_stereo
  Warning: Call to deprecated method fromCameraInfo. (The fromCameraInfo() method is deprecated as of J-turtle, and will be removed in K-turtle. Please use the from_camera_info() method instead.) -- Deprecated since version J-turtle.

test/test_directed.py: 192 warnings
  Warning: Call to deprecated method projectPixelTo3d. (The projectPixelTo3d() method is deprecated as of J-turtle, and will be removed in K-turtle. Please use the project_pixel_to_3d() method instead.) -- Deprecated since version J-turtle.

-- Docs: https://docs.pytest.org/en/stable/how-to/capture-warnings.html
---
Finished <<< camera_calibration [31.7s]

Summary: 1 package finished [32.1s]
  1 package had stderr output: camera_calibration

ScottMonaghan, 2024-04-20:
UPDATE: it looks like the updated image_geometry 4.1.0  package hasn't migrated yet to packages.ros.org, so I'll wait to make any changes to camera_calibration until the package has been officially updated.

[image]

Question: it looks like packages.ros.org was last updated at 11pm on Thursday 18-April. Is this something that happens weekly on Thursdays, or is there a different cadence?

ahcorde, 2024-04-22:
> Question: it looks like packages.ros.org was last updated at 11pm on Thursday 18-April. Is this something that happens weekly on Thursdays, or is there a different cadence?

ROS 2 releases are announced in ROS discourse and in general there is a 3-4 weeks cadence. You can take a look [here](https://discourse.ros.org/tag/release)

ScottMonaghan, 2024-04-22:
I have the fixes ready that are compatible with the released image_geometry 4.1 package, but the package isn't live on packages.ros.org yet, and the changes would be breaking otherwise. Should I create the PR or hold off until 4.1 is available through rosdep & apt?

ahcorde, 2024-04-22:
@ScottMonaghan you can create the PR, I will run CI using the sources.

ScottMonaghan, 2024-04-23:
Fix submitted in PR #968.

Once that is merged, this issue can be closed.

ScottMonaghan, 2024-04-23:
Now that #968 is merged. This can be closed.

closed 2024-04-24 by JWhitleyWork