ros-perception/image_pipeline #977: Topics do not appear to be synchronized

Rudresh172, 2024-05-01:
We are trying to use a stereo camera setup to generate pointcloud. This is our desired flow - 
[image]

We are using the image_proc node for rectification of images. We have created a custom publisher which subscribes to the /camera_info topic and publishes out the /camera_info_rect at the same publish rate (60) - 
```
public:
    CameraInfoPublisher() : Node("camera_info_publisher")
    {
        camera_ = this->declare_parameter("camera", "camera0");

        image_sub_ = this->create_subscription<sensor_msgs::msg::CameraInfo>("/olive/camera/olv_cam01/camera_info", 60,
              std::bind(&CameraInfoPublisher::image_callback_01, this, _1));

        image_sub2_ = this->create_subscription<sensor_msgs::msg::CameraInfo>("/olive/camera/olv_cam02/camera_info", 60,
              std::bind(&CameraInfoPublisher::image_callback_02, this, _1));

        publisher_ = this->create_publisher<sensor_msgs::msg::CameraInfo>("/olive/camera/olv_cam01/camera_info_rect", 60);
        publisher2_ = this->create_publisher<sensor_msgs::msg::CameraInfo>("/olive/camera/olv_cam02/camera_info_rect", 60); ///olive/camera/olv_cam02/camera_info_rect
    }
```
But it gives out this warning message in the terminal - 
```
[image_proc-7] [WARN] [1714602948.619293932] [rect_left]: [image_transport] Topics '/olive/camera/olv_cam01/image_raw' and '/camera_info' do not appear to be synchronized. In the last 10s:
[image_proc-7] 	Image messages received:      0
[image_proc-7] 	CameraInfo messages received: 36
[image_proc-7] 	Synchronized pairs:           0
[image_proc-6] [WARN] [1714602948.713715736] [rect_right]: [image_transport] Topics '/olive/camera/olv_cam02/image_raw' and '/camera_info' do not appear to be synchronized. In the last 10s:
[image_proc-6] 	Image messages received:      0
[image_proc-6] 	CameraInfo messages received: 38
[image_proc-6] 	Synchronized pairs:           0

``` 
Eventually, the pointcloud stops publishing (after around 2-3 minutes of use)
We want to know If this is happening because of the above warning?
Note that I am using a 40GB RAM i9 along with a RTX 3060, so we don't think compute is a problem here.

Thanks!

JWhitleyWork, 2024-05-01:
By default, nodes use a `SingleThreadedExecutor` which executes callbacks sequentially rather than in parallel. However, even a `MultiThreadedExecutor` would not really solve this issue. The solution is to use a `message_filters::Synchronizer` to time-synchronize the subscriptions. A basic example can be found here: https://answers.ros.org/question/366440/ros-2-message_filters-timesynchronizer-minimal-example-does-not-reach-callback-function/

Rudresh172, 2024-05-03:
Hello @JWhitleyWork,

Thanks a lot for your quick response. Do we need Time synchronizer since the ros2 camera plugin is publishing /image_raw and /camera_info msgs which are already synchronized? We are subscribing to camera01/camera_info and publishing camera01/camera_info_rect and similarly for camera02. This is our callback - 
```
private:
    void image_callback_01(sensor_msgs::msg::Image::SharedPtr img_msg)
    {
      // start with a blank message
      sensor_msgs::msg::CameraInfo info;
  
      path file_path = ament_index_cpp::get_package_share_directory("camera_intrinsics");
      //file_path /= path("config") / path(camera_ + "_cal") / path("ost.yaml");
      file_path /= path("config") / path("left_40.yaml");

      std::string saved_name; // camera name in file - to be loaded
     
      // parse the calibration into a CameraInfo message
      if (!camera_calibration_parsers::readCalibration(file_path, saved_name, info))
      {
        RCLCPP_INFO(this->get_logger(), "Error parsing calibration");
        return;
      }

      // need to fill in timestamp and frame info
      info.header = img_msg->header;

      //RCLCPP_INFO(this->get_logger(), "Publishing camera info");

      publisher_->publish(info);
    }

    void image_callback_02(sensor_msgs::msg::Image::SharedPtr img_msg)
    {
      // start with a blank message
      sensor_msgs::msg::CameraInfo info;

      path file_path = ament_index_cpp::get_package_share_directory("camera_intrinsics");
      file_path /= path("config") / path("right_40.yaml");

      std::string saved_name; // camera name in file - to be loaded
     
      // parse the calibration into a CameraInfo message
      if (!camera_calibration_parsers::readCalibration(file_path, saved_name, info))
      {
        RCLCPP_INFO(this->get_logger(), "Error parsing calibration");
        return;
      }

      // need to fill in timestamp and frame info
      info.header = img_msg->header;

      //RCLCPP_INFO(this->get_logger(), "Publishing camera info");

      publisher2_->publish(info);
    }

    rclcpp::Subscription<sensor_msgs::msg::Image>::SharedPtr image_sub_;
    rclcpp::Subscription<sensor_msgs::msg::Image>::SharedPtr image_sub2_;
    rclcpp::Publisher<sensor_msgs::msg::CameraInfo>::SharedPtr publisher_;
    rclcpp::Publisher<sensor_msgs::msg::CameraInfo>::SharedPtr publisher2_;
    std::string camera_;
    
};
```

We also tried subscribing to /image_raw instead of /camera_info since we are using the header directly from subscription, but it gives the same issue.
When we use the "stereo_image_proc.launch.py" and remap the topics, it works without issues. Is it because of the use of composable node and [GroupAction](https://github.com/ros-perception/image_pipeline/blob/171d4366a5ae5c70a0092654ed4bb6114f48ed48/stereo_image_proc/launch/stereo_image_proc.launch.py#L230)?

mikeferguson, 2024-05-30:
So I just scanned this issue - not digging super deep - but I'm going to to note it looks like you are using the "image_proc" node in ROS2 - which, basically has been horribly broken for a long time (I'm not sure it ever worked - it creates two instances of RectifyNode internally but doesn't set any names for them, so they have the same parameters). That node was actually removed in https://github.com/ros-perception/image_pipeline/pull/925 - and that PR also makes the image_proc.launch.py actually work (in porting the documentation, I tested all of these things).

Docs are now building here: https://docs.ros.org/en/rolling/p/image_proc/ - and there is a tutorial on using the launch file. I'm not sure which ROS2 distro you are using, some of the fixes haven't been ported all the way back

mikeferguson, 2024-10-29:
Closing due to lack of activity - please reopen if there is new information at a later date

closed 2024-10-29 by mikeferguson