ros-perception/image_pipeline #1021: image_view crashes with encoding yuv422_yuy2

christianrauch, 2024-08-07:
Using the `image_view` / `image_view::ImageViewNode` with image encoding `yuv422_yuy2` causes exception:
```
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.6.0) ./modules/imgproc/src/color.simd_helpers.hpp:92: error: (-2:Unspecified error) in function 'cv::impl::{anonymous}::CvtHelper<VScn, VDcn, VDepth, sizePolicy>::CvtHelper(cv::InputArray, cv::OutputArray, int) [with VScn = cv::impl::{anonymous}::Set<3, 4>; VDcn = cv::impl::{anonymous}::Set<3, 4>; VDepth = cv::impl::{anonymous}::Set<0, 2, 5>; cv::impl::{anonymous}::SizePolicy sizePolicy = cv::impl::<unnamed>::NONE; cv::InputArray = const cv::_InputArray&; cv::OutputArray = const cv::_OutputArray&]'
> Invalid number of channels in input image:
>     'VScn::contains(scn)'
> where
>     'scn' is 2
```

This is probably easiest reproducible with the `camera_ros` node:
```
# start camera with YUYV
ros2 run camera_ros camera_node --ros-args -p format:=YUYV
# start image view node
ros2 run image_view image_view --ros-args -r /image:=/camera/image_raw
```

The exception is thrown by `cv::imshow`.

quic-zhaoyuan, 2024-09-06:
Hi @christianrauch,
I have test the step on my laptop(ubuntu22.04, ros2 humble), seems it is working, cannot reproduce the issue you description
[image]

christianrauch, 2024-09-06:
> Is this issue based on specific Platform or ROS version

I tested this on Ubuntu 24.04 / ROS jazzy with and compiled the node from source.

christianrauch, 2024-09-08:
I can reproduce this also with the `v4l2_camera` package on `jazzy`:
```sh
# 'v4l2_camera' package, 'sudo apt install ros-jazzy-v4l2-camera'
ros2 run v4l2_camera v4l2_camera_node --ros-args -p output_encoding:=yuv422_yuy2
ros2 run image_view image_view --ros-args -r /image:=/image_raw
```

christianrauch, 2025-01-03:
@quic-zhaoyuan Can you reproduce with my instructions?

This is the full exception:
```
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.6.0) ./modules/imgproc/src/color.simd_helpers.hpp:92: error: (-2:Unspecified error) in function 'cv::impl::{anonymous}::CvtHelper<VScn, VDcn, VDepth, sizePolicy>::CvtHelper(cv::InputArray, cv::OutputArray, int) [with VScn = cv::impl::{anonymous}::Set<3, 4>; VDcn = cv::impl::{anonymous}::Set<3, 4>; VDepth = cv::impl::{anonymous}::Set<0, 2, 5>; cv::impl::{anonymous}::SizePolicy sizePolicy = cv::impl::<unnamed>::NONE; cv::InputArray = const cv::_InputArray&; cv::OutputArray = const cv::_OutputArray&]'
> Invalid number of channels in input image:
>     'VScn::contains(scn)'
> where
>     'scn' is 2
```

This is the full backtrace with all dbgsym packages:
```C++
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=<optimized out>, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff744526e in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff74288ff in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff78a5ff5 in __gnu_cxx::__verbose_terminate_handler () at ../../../../src/libstdc++-v3/libsupc++/vterminate.cc:95
#6  0x00007ffff78bb0da in __cxxabiv1::__terminate (handler=<optimized out>) at ../../../../src/libstdc++-v3/libsupc++/eh_terminate.cc:48
#7  0x00007ffff78a5a55 in std::terminate () at ../../../../src/libstdc++-v3/libsupc++/eh_terminate.cc:58
#8  0x00007ffff78bb391 in __cxxabiv1::__cxa_throw (obj=<optimized out>, tinfo=0x7ffff7369fe0 <typeinfo for cv::Exception>, dest=0x7ffff7213660 <cv::Exception::~Exception()>)
    at ../../../../src/libstdc++-v3/libsupc++/eh_throw.cc:98
#9  0x00007ffff706e7fa in cv::error (exc=...) at /usr/src/opencv-4.6.0+dfsg-13.1ubuntu1/modules/core/src/system.cpp:1243
#10 0x00007ffff7217e70 in cv::error
    (_code=-2, _err="Invalid number of channels in input image:\n    'VScn::contains(scn)'\nwhere\n    'scn' is 2", _func=<optimized out>, _file=<optimized out>, _line=92)
    at /usr/src/opencv-4.6.0+dfsg-13.1ubuntu1/modules/core/src/system.cpp:1255
#11 0x00007ffff70a8f63 in cv::detail::check_failed_auto_<int> (v=@0x7fffd4ffcd5c: 2, v@entry=@0x7fffd4ffcd5c: <optimized out>, ctx=...)
    at /usr/src/opencv-4.6.0+dfsg-13.1ubuntu1/modules/core/src/check.cpp:130
#12 0x00007ffff70a8fb8 in cv::detail::check_failed_auto (v=<optimized out>, ctx=...) at /usr/src/opencv-4.6.0+dfsg-13.1ubuntu1/modules/core/src/check.cpp:156
#13 0x00007ffff68dabb8 in cv::impl::(anonymous namespace)::CvtHelper<cv::impl::(anonymous namespace)::Set<3, 4, -1>, cv::impl::(anonymous namespace)::Set<3, 4, -1>, cv::impl::(anonymous namespace)::Set<0, 2, 5>, (cv::impl::<unnamed>::SizePolicy)2>::CvtHelper(cv::InputArray, cv::OutputArray, int)
    (this=this@entry=0x7fffd4ffcf10, _src=..., _dst=..., dcn=dcn@entry=3, this=<optimized out>, _src=<optimized out>, _dst=<optimized out>, dcn=<optimized out>)
    at /usr/src/opencv-4.6.0+dfsg-13.1ubuntu1/modules/imgproc/src/color.simd_helpers.hpp:92
#14 0x00007ffff68bb438 in cv::cvtColorBGR2BGR (swapb=true, dcn=3, _dst=..., _src=...) at /usr/src/opencv-4.6.0+dfsg-13.1ubuntu1/modules/imgproc/src/color_rgb.dispatch.cpp:552
#15 cv::cvtColor (_src=..., _dst=..., code=code@entry=4, dcn=3) at /usr/src/opencv-4.6.0+dfsg-13.1ubuntu1/modules/imgproc/src/color.cpp:198
#16 0x00007ffff776fe36 in convertToShow (toRGB=true, dst=..., src=...) at /usr/src/opencv-4.6.0+dfsg-13.1ubuntu1/modules/core/include/opencv2/core/mat.inl.hpp:92
#17 convertToShow (toRGB=true, arr=0x7fffa41833c0, src=...) at /usr/src/opencv-4.6.0+dfsg-13.1ubuntu1/modules/highgui/src/precomp.hpp:182
#18 DefaultViewPort::updateImage (this=0x7fffa4179f40, arr=<optimized out>) at /usr/src/opencv-4.6.0+dfsg-13.1ubuntu1/modules/highgui/src/window_QT.cpp:2645
#19 0x00007ffff7779a0d in CvWindow::updateImage (arr=0x7fffd4ffd420, this=<optimized out>) at /usr/src/opencv-4.6.0+dfsg-13.1ubuntu1/modules/highgui/src/window_QT.cpp:1876
#20 GuiReceiver::showImage(QString, void*) [clone .isra.0] (name=..., arr=arr@entry=0x7fffd4ffd420, this=<optimized out>)
    at /usr/src/opencv-4.6.0+dfsg-13.1ubuntu1/modules/highgui/src/window_QT.cpp:1101
#21 0x00007ffff7763cc7 in cvShowImage (name=<optimized out>, arr=0x7fffd4ffd420) at /usr/src/opencv-4.6.0+dfsg-13.1ubuntu1/modules/highgui/src/window_QT.cpp:773
#22 0x00007ffff7760465 in cv::imshow (winname="/image_raw", _img=...) at /usr/include/c++/13/bits/basic_string.h:222
#23 0x00007ffff7f53741 in image_view::ImageViewNode::windowThread (this=0x555555c42e80) at /usr/src/ros-jazzy-image-view-5.0.6-1noble.20241228.003136/src/image_view_node.cpp:291
#24 0x00007ffff78ecdb4 in std::execute_native_thread_routine (__p=0x555555f6db90) at ../../../../../src/libstdc++-v3/src/c++11/thread.cc:104
#25 0x00007ffff749ca94 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:447
#26 0x00007ffff7529c3c in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
```

quic-zhaoyuan, 2025-02-11:
Sure will do

zycczy, 2025-02-12:
I am quic-zhaoyuan with another github account, 
Has located the issue, the cv::imshow can only recevie brg/rgb, but this code [https://github.com/ros-perception/image_pipeline/blob/rolling/image_view/src/image_view_node.cpp#L217](url) let cv bridge out put yuv422, I have raised a PR(https://github.com/ros-perception/image_pipeline/pull/1071) to rolling to fix this issue.
Test result as below:
[image]

closed 2025-05-02 by christianrauch