ros-perception/image_pipeline #1022: RectifyNode gray calibration

kbiesiadecki141, 2024-08-15:
**Version**: ROS2 Iron (Ubuntu 22.04)

**Problem:** Image rectification via RectifyNode is incorrect

**Observations:** I tried two different methods for applying a calibration to an image:

1. The one shown below on the left uses Foxglove Studio, which allows you to select an Image topic and optionally apply a calibration by specifying a calibration URL. This rectification appears as expected given the input calibration file. 
2. The one shown below on the right uses the`image_proc::RectifyNode` . This is the desired method, as it would produce the rectified image topic required for `stereo_image_proc`.

I am quite certain that these have the same raw image input and calibration input.

[image]

Any support would be greatly appreciated.

mikeferguson, 2024-08-16:
Is it possible to post a short bagfile (a single camera + info topic would be fine) along with the configuration you're using with RectifyNode? I have been using the RectifyNode recently on my robot and it is working - so I'm thinking this is some configuration issue (there's a lot of redundant data in CameraInfo, so it's possible that Foxglove uses different parts of the message than image_pipeline does)

kbiesiadecki141, 2024-08-16:
Here's a short camera and info bag: [rosbag2_2024_08_16-18_28_17_0.zip](https://github.com/user-attachments/files/16640489/rosbag2_2024_08_16-18_28_17_0.zip)

I don't have any parameters set for RectifyNode, so it should be using its default configuration. The launch invocation is shown below:
[image]

mikeferguson, 2024-08-19:
I'm not sure how you calibrated your camera - but I think the P and R matrices are pretty messed up - my guess here is that Foxglove is only using the K matrix, whereas image_pipeline/image_geometry/openCV are using the P & R matrices:

```
k:
- 771.2541758055805  // fx
- 0.0
- 941.6836266931062  // cx
- 0.0
- 771.2541758055805  // fy
- 601.6937980646015  // cy
- 0.0
- 0.0
- 1.0
p:
- 789.5162907369597  // fx (different, but at least somewhat close to k matrix)
- 12.174608855352242  // should be nearly 0?
- 0.019239452452525115  // cx - this seems WAY off as it is nowhere close to the center of the width
- 0.0
- -0.18066087146710358  // should be nearly 0
- 771.5852315230734   // fy (nearly same as K matrix)
- 0.00031072349544309096  // cy - again, seems way off
- 0.0
- 934.2211723150673
- 607.9561467627569
- 0.9998148563210273
- 0.0
```

If I set P[1:3,1:3] = K, and clear the R matrix to be an identity matrix, your images come through rectification from the bagfile, and they have a small bit of correction applied (so they are getting rectified by falling back to the proper K matrix).

mikeferguson, 2024-08-23:
I'm going to close this as likely not an issue with image pipeline - please reopen if you end up having more information that indicates an actual bug.

closed 2024-08-23 by mikeferguson

acarrillo, 2024-08-27:
> I'm going to close this as likely not an issue with image pipeline - please reopen if you end up having more information that indicates an actual bug.

Thanks Mike, I concur with you that there is something up with our `CameraInfo` data population. Really appreciate your help. I will let you know if we end up finding something otherwise!