ros-perception/image_pipeline #1026: Adding new Parameter for PointCloudXyz...Node of package depth_image_proc

HappySamuel, 2024-08-23:
Hi

Is it possible to add new parameter for relevant `PointCloudXyz....Node`
`skip: Skips **skip** number of depth pixels in order to limit the number of pixels converted to points`
With this parameter, we are able to limit number of pixels converted to pointcloud, hence reduce the computation further down stream.

Best,
Samuel

mikeferguson, 2024-08-23:
Seems like a useful feature - a PR would be accepted.

I think most people currently do this using CropDecimate on the image and then running the decimated image into the PointCloud node. It might make sense to actually use the same parameter names "decimation_x" and "decimation_y" rather than "skip" so that you could independently control downscaling in the two axis of the depth image.

muhd360, 2024-09-01:
i  would like to be assigned this issue

muhd360, 2024-09-09:
hi any updates @mikeferguson