ros-perception/image_pipeline #1028: Change callback group

TheDelus, 2024-08-29:
There are some nodes (eg. image_proc/debayer) that could benefit from changing its default callback group from MutuallyExclusive to Reentrant. Is there a way to do this?

mikeferguson, 2024-08-30:
Each of these are designed as a ROS2 component - it's my understanding that components are loaded with a single threaded executor, so I don't think changing the callback groups would do anything at this time?

TheDelus, 2024-08-30:
As far as I know this depends on the used `component_container`. I am using the [component_container_mt](https://github.com/ros2/rclcpp/blob/rolling/rclcpp_components/src/component_container_mt.cpp). I have created a local fork of the `image_proc/debayer` node and changed the callback group of the subscriber to Reentrant. It is running perfectly fine in parallel now. Would be nice to configure stuff like this.

mikeferguson, 2024-08-30:
Wasn't aware of the alternate containers - that's cool. If you already have an example - please open a PR