ros-perception/image_pipeline #1029: [image_saver_node]: Unable to convert 8UC3 image to bgr8

HaiCLi, 2024-09-04:
Hi guys,

I met this problem when I try to save the images from rosbag2 with timestamp. Did anyone knows how to solve it?

Thanks!

 
[image]

HaiCLi, 2024-09-04:
"ros2 run image_view image_saver --encoding=8Uc3 --ros-args -r image:=/cam/video0 " the command I used

mikeferguson, 2024-09-04:
It would appear to me that you are not passing the parameters correctly since it is still using the default "bgr8":

     ros2 run image_view image_saver --ros-args -p encoding:=8UC3 -r image:=/cam/video0

HaiCLi, 2024-09-05:
> It would appear to me that you are not passing the parameters correctly since it is still using the default "bgr8":
> 
> ```
>  ros2 run image_view image_saver --ros-args -p encoding:=8UC3 -r image:=/cam/video0
> ```

could you provide the correct one? I can not find it on RosWiki. Thanks!

HaiCLi, 2024-09-05:
> The command I posted would be the correct one - you move the parameter after --ros-args and add the -p as shown

Thanks for your reply! However, I found the results are lack of timestamps. I need this attribution for the use of sensor fusion with lidar. Do you have any suggestions? Thanks!