ros-perception/image_pipeline #1042: New publisher discovered on topic 'left/right', offering incompatible QoS. No messages will be received from it. Last incompatible policy: RELIABILITY

l4es, 2024-11-12:
Hi,

I attempt to perform the stereo camera calibration for my own stereo camera : 

```
$ ros2 topic list
/baro/pressure
/camera/assemble/image_raw
/camera/assemble/image_raw/compressed
/camera/left/camera_info
/camera/left/image_raw
/camera/left/image_raw/compressed
/camera/right/camera_info
/camera/right/image_raw
/camera/right/image_raw/compressed
```

by launching the following command : 
`$ ros2 run camera_calibration cameracalibrator --approximate 0.1 --size 8x6 --square 0.108 right:=/camera/right/image_raw left:=/camera/left/image_raw right_camera:=/camera/right left_camera:=/camera/left`

which gave the following log : 
```
[WARN] [1731403429.652019518] [rcl]: Found remap rule 'right:=/camera/right/image_raw'. This syntax is deprecated. Use '--ros-args --remap right:=/camera/right/image_raw' instead.
[WARN] [1731403429.652053407] [rcl]: Found remap rule 'left:=/camera/left/image_raw'. This syntax is deprecated. Use '--ros-args --remap left:=/camera/left/image_raw' instead.
[WARN] [1731403429.652066111] [rcl]: Found remap rule 'right_camera:=/camera/right'. This syntax is deprecated. Use '--ros-args --remap right_camera:=/camera/right' instead.
[WARN] [1731403429.652075232] [rcl]: Found remap rule 'left_camera:=/camera/left'. This syntax is deprecated. Use '--ros-args --remap left_camera:=/camera/left' instead.
[WARN] [1731403429.660325344] [rcl]: Found remap rule 'right:=/camera/right/image_raw'. This syntax is deprecated. Use '--ros-args --remap right:=/camera/right/image_raw' instead.
[WARN] [1731403429.660359490] [rcl]: Found remap rule 'left:=/camera/left/image_raw'. This syntax is deprecated. Use '--ros-args --remap left:=/camera/left/image_raw' instead.
[WARN] [1731403429.660370786] [rcl]: Found remap rule 'right_camera:=/camera/right'. This syntax is deprecated. Use '--ros-args --remap right_camera:=/camera/right' instead.
[WARN] [1731403429.660380482] [rcl]: Found remap rule 'left_camera:=/camera/left'. This syntax is deprecated. Use '--ros-args --remap left_camera:=/camera/left' instead.
Waiting for service camera/set_camera_info ...
OK
Waiting for service left_camera/set_camera_info ...
OK
Waiting for service right_camera/set_camera_info ...
OK
[WARN] [1731403444.972188340] [cameracalibrator]: New publisher discovered on topic 'left', offering incompatible QoS. No messages will be received from it. Last incompatible policy: RELIABILITY
[WARN] [1731403444.973163030] [cameracalibrator]: New publisher discovered on topic 'right', offering incompatible QoS. No messages will be received from it. Last incompatible policy: RELIABILITY
```
But nothing appeared on the GUI despite that the checkerboard did appear on rqt for both /camera/left/image_raw /camera/right/image_raw topics.

Could someone help, please ?
Khang

mikeferguson, 2024-11-12:
Please post an image of what the GUI looks like, and also check the terminal for any logged errors

l4es, 2024-11-12:
Hi @mikeferguson,

Sorry for not providing more detail. Please find my answer inline 

RQT shows checkerboard inside of both frames : 
[image]

> an image of what the GUI looks like

[image]

[image]

> check the terminal for any logged errors and post them here

```
~$ ros2 run camera_calibration cameracalibrator --approximate 0.1 --size 8x6 --square 0.108 --ros-args --remap right:=/stereo/right/image_raw --remap left:=/camera/left/image_raw --remap right_camera:=/camera/right --remap left_camera:=/camera/left
Waiting for service camera/set_camera_info ...
OK
Waiting for service left_camera/set_camera_info ...
OK
Waiting for service right_camera/set_camera_info ...
OK
```
> note which ROS distribution you are using, and whether from source or pre-built Debian packages
ROS2 Humble installed from pre-built .deb packages with following reference : https://qiita.com/ntrlmt/items/50b49fb43b4439ac165b

Potentially, the image_pipeline for Nvidia Jetson/Jetpack could be this one  : https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_image_pipeline

Thanks,
K

l4es, 2024-11-12:
Hi @mikeferguson,

Thanks very much. The checkerboard is now detected.
[image]
 By the way, for the FoV 185 (D) x 160 (H) x 98 (V),  what should I select the camera model? Should it be fisheye rather than pinhole ? And what is the purpose of the scale, please ? 

Thanks,
Khang

mikeferguson, 2024-11-12:
Looks like you're up and running if you switch to regular pinhole - I'm not sure fisheye calibration actually works in Humble - I've personally never used fisheye (in any distro)