ros-perception/image_pipeline #1056: "commit" crashes the camera calibrator

rursprung, 2024-12-04:
it seems that in the camera calibrator the "commit" button is broken: when i do a calibration and click "calibrate", then "save" these two work. if i follow that with "commit" it crashes every time:

```
Traceback (most recent call last):
  File ".../ros_ws/install/camera_calibration/lib/camera_calibration/cameracalibrator", line 33, in <module>
    sys.exit(load_entry_point('camera-calibration==5.0.5', 'console_scripts', 'cameracalibrator')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File ".../ros_ws/install/camera_calibration/lib/python3.12/site-packages/camera_calibration/nodes/cameracalibrator.py", line 227, in main
    rclpy.shutdown()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/__init__.py", line 134, in shutdown
    _shutdown(context=context)
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/utilities.py", line 82, in shutdown
    context.shutdown()
  File "/opt/ros/jazzy/lib/python3.12/site-packages/rclpy/context.py", line 127, in shutdown
    raise RuntimeError('Context must be initialized before it can be shutdown')
RuntimeError: Context must be initialized before it can be shutdown
[ros2run]: Process exited with failure 1
```

tested on ROS jazzy with a local copy of this repo in my ROS workspace with 5.0.5 + a cherry-pick of my fix #1046 (i.e. for the bug reported here it's identical to 5.0.5).

reubendocea, 2025-01-21:
Likewise. Any update?

fallenmi, 2026-08-15:
Thanks for the reproducible traceback. I’m working on a focused fix that makes the outer shutdown idempotent and adds lifecycle tests for both the Commit path and a normal UI exit.

closed 2026-08-27 by ahcorde