ros-perception/image_pipeline #1060: image_view causes 100% CPU usage

rursprung, 2024-12-22:
when i run `image_view` (using `ros2 run image_view image_view --ros-args -r image:=/my/image`) for 1-2 topics then the CPU usage goes up to 100%. i've tested this on several devices, all with the same result.

this is with ROS jazzy, image-view version 5.0.5 from apt (5.0.6 hasn't been synced to apt yet, but based on the commits it doesn't include anything relevant for this)

rursprung, 2024-12-28:
see this screenshot of `htop` - this is running in a WSL2 Ubuntu VM and you can see that i'm running `image_view` 4x (to see 4 different topics) and each one uses 100% of a CPU core:
[image]

this is at a time where there aren't even any new messages to be processed (i was playing a ROS bag but already stopped `ros2 bag play` but hadn't stopped the `image_view` nodes yet)

mikeferguson, 2025-02-21:
I think this is because if there is no new image, the window thread never sleeps - I created a possible fix in #1082