ros-perception/image_pipeline #1064: Camera, left_camera and right_camera parameters aren't used

PfeifferMicha, 2025-02-04:
I can no longer change where the camera_calibrator looks for the set_camera_info parameters, because paths are hard coded.

I might be missing something, but it looks like currently on `rolling`, `jazzy` and `humble` branches, the camera_calibrator ignores the camera, left_camera and right_camera parameters (again).

This **was** already reported in https://github.com/ros-perception/image_pipeline/issues/754 and fixed in:
https://github.com/ros-perception/image_pipeline/pull/755
It had also already been placed on rolling:
https://github.com/ros-perception/image_pipeline/pull/761

But somehow the tip of the all of current branches is missing this change, making it impossible to set the location of the set_camera_info services. They all contain hard-coded paths:
https://github.com/ros-perception/image_pipeline/blob/722ca08b98f37b7b148d429753da133ff1e2c7cf/camera_calibration/src/camera_calibration/camera_calibrator.py#L106

Or am I missing something and there's now a different way to remap the service locations?

mikeferguson, 2025-02-04:
It looks like https://github.com/ros-perception/image_pipeline/pull/792 reverted the commit you mentioned, in favor of using remap rules rather than parameters (which is more ROS-y). You should be able to use a remap rather than a parameter - for instance, the linked PR has the following example:

Per our [documentation](https://docs.ros.org/en/rolling/p/camera_calibration/doc/tutorial_stereo.html), you should be able to remap the left/right cameras:

```
ros2 run camera_calibration cameracalibrator --approximate 0.1 --size 8x6 --square 0.108 right:=/stereo/right/image_raw left:=/stereo/left/image_raw right_camera:=/stereo/right left_camera:=/my_stereo/left
```

The first remap is for the image_raw/camera_info topics, and the second is for the camera info services. As noted in that tutorial, that command assumes your topic/service names are:

```
/stereo/left/camera_info
/stereo/left/image_raw
/stereo/left/set_camera_info (service)
/stereo/right/camera_info
/stereo/right/image_raw
/stereo/right/set_camera_info (service)
```

PfeifferMicha, 2025-02-04:
Ah, thanks!

I had already tried that, though. I just get:

```

Waiting for service camera/set_camera_info ...
Service camera/set_camera_info not found.
Waiting for service left_camera/set_camera_info ...
Service left_camera/set_camera_info not found.
Waiting for service right_camera/set_camera_info ...
Service right_camera/set_camera_info not found.
```
followed by a:
```
[...]ROS2/ws/build/camera_calibration/src/camera_calibration/camera_calibrator.py", line 125, in __init__
    raise RuntimeError("no camera service available")
```

I tried stereo calibration, but also monocular calibration doesn't work (throws the same errors):
`ros2 run camera_calibration cameracalibrator --approximate 0.1 --size 8x6 --square 0.108 image:=/stereo/right_fullscale/image_raw camera:=/stereo/split_node_container`

Note that these two services do exists. Shouldn't that be enough?
```
/stereo/split_node_container/get_type_description
/stereo/split_node_container/set_camera_info
```

mikeferguson, 2025-02-04:
Several followups:
 * What ROS version are you using?
 * Can you try the more explicit remapping command (I know the older syntax has been deprecated for a bit, maybe it has stopped working altogether): ```ros2 run camera_calibration cameracalibrator --approximate 0.1 --size 8x6 --square 0.108 --ros-args -r right:=/stereo/right/image_raw -r left:=/stereo/left/image_raw -r right_camera:=/stereo/right -r left_camera:=/my_stereo/left```

PfeifferMicha, 2025-02-04:
Sorry, should have mentioned both of those things:

- ROS jazzy, Ubuntu 24.04. I'm also on the jazzy branch of this repo.
- Sorry, yes, I had also already tried the new-style remapping, for example:

`ros2 run camera_calibration cameracalibrator --approximate 0.1 --pattern charuco --size 10x14 --aruco_dict 4x4_250 --charuco_marker_size 0.0105 --square 0.01693 --ros-args -r right:=/stereo/right_fullscale/image_raw -r left:=/stereo/left_fullscale/image_raw -r right_camera:=/stereo/split_node_container -r left_camera:=/stereo/split_node_container`

This exact command works when I cherry-pick the commit which uses parameters and change the last two -r to -p.
(I realize I use the same camera twice, that's on purpose and due to [another, probably unrelated issue](https://github.com/ros-perception/image_common/issues/341) that I have with creating two CameraInfoManagers... this is why I've also tried with just one camera above).

I'll keep digging and try to build a minimal example. Maybe there's an issue with the service, because I create it inside a composable node and have little experience with those...

PfeifferMicha, 2025-02-04:
Ah, I found the solution!

Remapping like this doesn't work:
`-r left_camera:=/stereo/split_node_container`
but remapping like this works:
`-r left_camera/set_camera_info:=/stereo/split_node_container/set_camera_info`

This works:

`ros2 run camera_calibration cameracalibrator --approximate 0.1 --pattern charuco --size 10x14 --aruco_dict 4x4_250 --charuco_marker_size 0.0105 --square 0.01693 --ros-args -r right:=/stereo/right_fullscale/image_raw -r left:=/stereo/left_fullscale/image_raw -r right_camera/set_camera_info:=/stereo/split_node_container/set_camera_info -r left_camera/set_camera_info:=/stereo/split_node_container/set_camera_info`

But why...?

mikeferguson, 2025-02-04:
So, I'm not very read up on rclpy - but it looks like there is a ```resolve_service_name``` function (which is discussed in https://github.com/ros2/rclpy/issues/954 - and that's literally the only reference I can find). I know in C++ we have to do ```resolve_topic_or_service_name``` for remapped namespaces to work properly for compressed topics - but maybe it has less to do with compressed topics versus just namespaces?