ros-perception/image_pipeline #1069: Camera calibrator not recognizing the checkerboard

fabiocyro, 2025-02-07:
Hello there, I'm trying to calibrate a raspberry pi camera with ROS2 humble.

After managing to make everything work, I'm now trying to calibrate the camera.

The problem is that it doesn't matter the size, light conditions, how close or how far I am from the camera it just does not detect, so I can't calibrate.

Any suggestions would be greatly appreciated

mikeferguson, 2025-02-12:
Most of the time, when this happens, it is because your checkerboard size is entered wrong - it's the number of intersections, not the number of blocks. For instance, below is an 8x6 checkerboard:

[image]

fabiocyro, 2025-02-12:
Thank you mike, it was indeed this the error.