ros-perception/image_pipeline #1085: fails to bulild Humble branch on ubuntu 20.04

andanders, 2025-04-03:
I wish to build image_pipeline so I can use `image_view` on ubuntu 20.04, and Humble that was build from source. Steps to reproduce:

`Git clone https://github.com/ros-perception/image_pipeline.git`
`git checkout humble`
from `workspace/` `colcon build --symlink-install`

Result 

```bash
colcon build --symlink-install
Starting >>> cv_bridge
Starting >>> image_geometry              
Starting >>> tracetools_image_pipeline
Finished <<< tracetools_image_pipeline [0.90s]                                                                                         
[Processing: cv_bridge, image_geometry]                                                          
Finished <<< image_geometry [34.1s]                                                           
[Processing: cv_bridge]                                      
Finished <<< cv_bridge [1min 9s]                              
Starting >>> image_proc     
Starting >>> camera_calibration
Starting >>> depth_image_proc                                                           
Starting >>> image_publisher
Starting >>> image_rotate                                                                                                                  
Starting >>> image_view                                                                                                                                          
--- stderr: image_publisher                                                                                                                                                                                  
/home/nvidia/humble_ws/src/image_pipeline/image_publisher/src/image_publisher.cpp:41:10: fatal error: cv_bridge/cv_bridge.h: No such file or directory
   41 | #include "cv_bridge/cv_bridge.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/image_publisher.dir/build.make:63: CMakeFiles/image_publisher.dir/src/image_publisher.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:107: CMakeFiles/image_publisher.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< image_publisher [2.90s, exited with code 2]
Aborted  <<< image_view [2.78s]
Aborted  <<< camera_calibration [3.04s]
Aborted  <<< image_rotate [5.47s]                                                                                                                    
Aborted  <<< image_proc [23.0s]                                                                                        
Aborted  <<< depth_image_proc [24.9s]                                       

Summary: 3 packages finished [1min 36s]
  1 package failed: image_publisher
  5 packages aborted: camera_calibration depth_image_proc image_proc image_rotate image_view
  4 packages had stderr output: depth_image_proc image_proc image_publisher image_rotate
  5 packages not processed

```

mikeferguson, 2025-04-03:
Humble is not supported on 20.04 - you should use Ubuntu 22.04

andanders, 2025-04-03:
I see. But the package is available to all the previous ROS distributions on Ubuntu Focal. And Humble has tier 3 support on Focal. Would it be feasible to hack out a variant of the image_view package that runs on humble on 20.04?

mikeferguson, 2025-04-03:
> I see. But the package is available to all the previous ROS distributions on Ubuntu Focal. 

Focal would have used the 'foxy' branch - which is still available, but no longer supported.

> And Humble has tier 3 support on Focal.

From [REP-2000](https://www.ros.org/reps/rep-2000.html) - this doesn't mean it is actually supported by developers:

```
Tier 3 platforms are those for which community reports indicate that the release is functional. The development team does not run the unit test suite or perform any other tests on platforms in Tier 3. Installation instructions should be available and up-to-date in order for a platform to be listed in this category. Community members may provide assistance with these platforms.
```

> Would it be feasible to hack out a variant of the image_view package that runs on humble on 20.04?

Maybe? But maintainers don't have the resources to help with this. I'd suggest you instead post to robotics.stackexchange.com and see if you can find help there.