ros-perception/image_pipeline #1091: Change QoS settings for published topics on Humble

Alessio-Parmeggiani, 2025-04-30:
hi, I would like to know if there is a way to change the QoS settings for published topics.
In particular I am using depth_image_proc::point_cloud_xyzrgb_node, and I am interested in changing the reliability for the published topic from BEST_EFFORT to RELIABLE from a launch file.

I noticed that this feature should be available in Jazzy: #847 , it is available also in Humble? If not, it is planned to be backported?

mikeferguson, 2025-04-30:
I don't think this is currently available in Humble currently. Our codebase has diverged quite a bit during the Jazzy transition as we were finally able to implement lazy publishers and did a massive cleanup to make the components all work in a similar fashion (a fairly large, but probably not entirely complete list is available here: https://docs.ros.org/en/rolling/p/image_pipeline/changelog.html)

Most of these changes were not backported since there was concern around breaking existing users of Humble (many of these changes couldn't really be done without changing the default behavior). I'm also not sure the QoS override API actually exists on Humble - I think that may have been added to rclcpp in a later release.

If you find that the change can be backported in a way that doesn't change default behavior, a PR would be accepted - but the maintainers definitely don't have the time to do investigate such a change.

closed 2025-04-30 by mikeferguson

Alessio-Parmeggiani, 2025-05-01:
Thank you for the answer! I understand the situation, I'll try to investigate a bit if this is possible in Humble